ryan-brazeal-ufl/OpenPyLivox #1: The little girl at home page really scares me

LitoNeo, 2019-09-27:
I notice at the README that it offers some funcions that LIVOX_VIEWER does, so I wonder if there is a method or API that I can use to handle .lvx files?

BTW, thanks for this awesome project, it's really helpful !
ewwww, yeah and I am still scared and it would be a long night tonight.....

ryan-brazeal-ufl, 2019-10-01:
Are you wanting for OpenPyLivox to be able to create a .LVX file? For playback/usage within the Livox-Viewer? I am aware of the Livox documentation on their LVX format and will gladly add support for creating LVX files as an output from OpenPyLivox. In an upcoming release there will be support for the ASPRS .LAS v1.2 file format.

LitoNeo, 2019-10-05:
Yes，thanks！

ryan-brazeal-ufl, 2020-09-11:
Hello,
OpenPyLivox (OPL) has just been updated to v1.1.0 and now fully supports the Livox Horizon sensor. A fellow developer has expressed interest in contributing an LVX reader to OPL, stay tuned for updates.
All the best,
-RB

closed 2020-09-11 by ryan-brazeal-ufl