ryan-brazeal-ufl/OpenPyLivox #4: Using with Jetson Nano

crea397, 2020-04-16:
Thank you for a great project!
Can I use the OpenPyLivox from the Jetson Nano?
Thanks.

ryan-brazeal-ufl, 2020-04-20:
Hello,
Unfortunately I have no experience using the Jetson Nano so I can't directly comment on using OpenPyLivox on the device. However, the Nano has an ethernet port and fully supports Python 3 so that's all you need for connecting to a Mid-40 sensor and using OpenPyLivox.
Sorry I couldn't be more help. Please let me know if you have any other more specific questions. Glad to help!
-RB

crea397, 2020-04-20:
Hi,
Thank you for your polite reply.
I'm going to try to see if I can use OpenPyLivox with the Jetson Nano.
I'll report back on the results of my attempts.

crea397, 2020-05-30:
Hi, 
I was able to run OpenPyLivox successfully on JetsonNano. Thank you very much!

closed 2020-06-04 by ryan-brazeal-ufl