ryan-brazeal-ufl/OpenPyLivox #5: problem message with crcmod

fabricedu41, 2020-05-03:
Hello, thank you for your project!
I have a problem when I run the demo script livox controller , I get the message "ModuleNotFoundError: No module named 'crcmod' in line 34 of file openpilivox.
I installed crcmod 1.7 I also did the crcmod test but I don't understand? could you guide me?
I read the wiki and installed openpilivox as directed.

config python 3.7
window 10 64 bit 
lidar mdr40 
crcmod installed 1.7 

Thank for your help :)

fab

fabricedu41, 2020-05-04:
Hi my problem was to interpret it spyder which I used to launch the script.
I tested directly from python 3.7 and it works !! it's fantastic !
I will now test with the different firmware :)!

ryan-brazeal-ufl, 2020-05-06:
Thanks for continuing to dig into your problem, solving it, and reporting the solution here. Much appreciated!

closed 2020-05-06 by ryan-brazeal-ufl