ryan-brazeal-ufl/OpenPyLivox #7: Real-time processing of point cloud data

crea397, 2020-05-30:
I want to perform real time object detection by processing the point cloud data that the Livox Mid-40 is sensing in real time.

The sample program of OpenPyLivox captures the streamed point cloud data by using saveDataToFile() and saveDataToCSV(), and then writes it to a binary file or CSV, and saves it at the end.

I want to process the point cloud data (e.g., data sensed in one second) that is being sensed in real time from the Mid-40, using a point cloud processing library such as Open3D, and perform object detection in real time.

I think I probably need something like batch processing.

In case of batch processing, it is necessary to save one second of point cloud data from the Mid-40 in real time to a CSV file, read the point cloud data from the CSV file, and perform the point cloud processing.

Do you have any good and efficient ideas for using OpenPyLivox?

ryan-brazeal-ufl, 2020-06-04:
Hello,
Thanks for the inquiry. In an upcoming release (I promise very soon) the ability to capture and store in memory a block for point observations for a small time period will be implemented. The block will be overwritten at the start of the next time period, so memory usage will remain small. The idea is similar to the Livox Viewer's real-time display options for 100ms, 250ms, 500ms, etc. I will be sure to add a 1s time interval as well. A new object property will be available that provides access to the block of point observations, and probably some block ID property so a user can determine when a new block is available. This ability will work as a standalone feature as well as when recording the point observations to a file.
Thanks again,
Ryan

crea397, 2020-06-05:
Hi Ryan.

Thank you for your polite reply!
I'm looking forward to the implementation of the new feature.

roya-correct-ai, 2020-07-12:
> Hi Ryan.
> 
> Thank you for your polite reply!
> I'm looking forward to the implementation of the new feature.

Hello
Could you please let me know whether you find a solution to have real-time processing on the lidar data? 
I appreciate it if you share your experience and progress on this issue.
Many thanks in advance

Shibaditya99, 2022-03-21:
so is it available? The realtime processing of the livox data? @ryan-brazeal-ufl

tensorturtle, 2022-05-30:
Hi @ryan-brazeal-ufl , I'm really looking forward to real-time processing feature!

zapplelove, 2022-10-27:
@ryan-brazeal-ufl @roya-correct-ai @crea397 @tensorturtle @Shibaditya99 I'm really looking forward to real-time processing feature too! I want to convert the stream to numpy array, and to deal with it in real time to realize the obstacle avoidance.