ryan-brazeal-ufl/OpenPyLivox #8: How do I capture the IMU data? and when the new version of the library will be released?

roya-correct-ai, 2020-07-10:
Hi Ryan, 

Thank you for this library. It helps me a lot.
Actually, I have 2 problems. The first one is that with the livox_controller_demo.py I can't have a real-time process on the lidar data. I saw you said that in the new version you will implement the real-time processing to the library. 
could you please let me know when you can release it?
The second request, I do not know how I can capture the IMU data by the OpenPyLivox library. 
I appreciate it if you help me with this issue.

Many thanks in advance
Roya

ryan-brazeal-ufl, 2020-07-17:
Hi Roya,
Thanks for posting your questions. Unfortunately I haven't yet started developing OPL to support the latest firmware(s) and the latest sensors. In an upcoming release support for the Horizon sensor will be included. I'm aware of the additional sensors (and observations) within the Horizon, and I will be supporting as many of the SDK functions for the Horizon sensor as possible. I don't currently have an estimated time on when this new release will be ready.
Thanks,
Ryan

roya-correct-ai, 2020-07-17:
Hi Ryan,

Thank you for your response. I am looking forward for the new version. 
About IMU data, would you please guide me on how I can capture it? I mean, with the available OPL library, is it possible to change any function or method to can capture the IMU data? is it the same as XYZ data? what is the difference? 
I would appreciate it if you help me because I have no idea how it is possible.

Many thanks in advance
Roya

ryan-brazeal-ufl, 2020-09-11:
Hello,
OpenPyLivox (OPL) has just been updated to v1.1.0 and now fully supports the Livox Horizon sensor, including IMU data.
All the best,
-RB

closed 2020-09-11 by ryan-brazeal-ufl