ryan-brazeal-ufl/OpenPyLivox #10: LVX File Reader

mar2194, 2020-07-17:
Hi Ryan,
I am currently mentoring a high school student who is helping with the development of a UAV-LIDAR mapping system that uses the Livox Horizon LIDAR. She has written a fantastic .lvx file reader in Python and I was hoping she could contribute the file reader to your repository somehow.

ryan-brazeal-ufl, 2020-07-17:
Hello,
Thanks for reaching out! I would be honoured to include your student's LVX file reader within OPL, with all the credit going to her/and her team! Kudos to your student for her time and effort dealing with the LVX file format, that's no easy task. I am a father of 3 girls (2 of which are in high school), so your student's software development goals are very meaningful to me as well. If there is anything I can do to assist/support her project(s), please don't hesitate to ask. As a starting point, may I ask that you email me at ryan dot brazeal at ufl dot edu, and we can figure out the specifics. Are you familiar with Git and the concept of Pull Requests? If not, I would recommend you read a little about it.
Thanks again,
Ryan

ryan-brazeal-ufl, 2020-09-11:
Hello,
My sincere apologies for the delay in getting back to you and more importantly, to your student. OpenPyLivox (OPL) has just been updated to v1.1.0 and now more-or-less fully supports the Livox Horizon sensor. Please check it out (upgrade instructions can be found in the Wiki) and report back with any new issues, questions, bugs, etc. As we briefly mentioned before, I would encourage your student to fork the repo., review it, email me with questions (I promise I will reply!!), and inject her own .lvx reader code where she thinks it best fits within OPL. Then submit a PR.
Many thanks,
-RB

mar2194, 2020-09-11:
Awesome Ryan. I'm having her do it now...

closed 2020-09-24 by ryan-brazeal-ufl