ryan-brazeal-ufl/OpenPyLivox #15: new livox LIDAR AVIA

fabricedu41, 2020-11-06:
Hi Ryan, do you think openpylivox can also be compatible with the new livox AVIA to get point cloud and x, y, z from imu?
With these data, it would be possible to integrate it on a drone to carry out lidar measurements .... great!

thank you :)
fab

ryan-brazeal-ufl, 2020-11-06:
Hi Fab,

Though I don't have (nor will I have at least for the foreseeable future) an Avia sensor, I quickly glanced at the updated Livox SDK information and it appears that OpenPyLivox (OPL) is already able to support the Avia sensor, though only using single or dual return mode. The triple return mode utilizes a **new data_type 7 and 8** ([read about it here](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol#33-point-cloudimu-data-data_type)) and this is **not** currently supported in OPL v1.1, but I will commit an updated OPL version soon. Also, the Avia supports either a non-repetitive scan pattern (like all other Livox sensors) or a new repetitive scan pattern, and currently OPL v1.1 does **not** support commanding the sensor to change scan patterns. However, before I can implement full support for the Avia sensor I have to wait for Livox to reply to the new issue I have logged with them, as their SDK documentation does not even state how to command the Avia sensor to utilize triple return mode (only up to dual return mode commands are listed in their documentation)? [see the command 0x06](https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol#0x06-set-lidar-return-mode).

OPL v1.1 already supports receiving the 200Hz IMU data for the Horizon and Tele sensors (and now also the Avia) but the observations are not x, y, z positions, rather, accelerations and rotation rates (like all other IMUs). The short answer is, IMUs are traditionally integrated, via what's called loose coupling or tight coupling, with a GNSS receiver. In such an integrated sensor the GNSS receiver provides the x, y, z positions while the IMU provides the roll, pitch, heading orientation angles (this is a huge over-simplification of what is actually happening, but it helps to think about it this way). For completeness, you can use the IMU-only data to estimate positional changes, via what's called dead reckoning, but the biases within the IMU sensor will cause these positional change estimates to degrade very quickly with time. I hope this helps you understand IMUs a little better?

Thanks,

Ryan

fabricedu41, 2020-11-07:
Hello Ryan :)

Thank you very much for your details, it's very exciting !

I believed that the new  AVIA lidars made it possible to have more precise IMU data for use directly on UAV....
It is therefore necessary to add a 'precision' 'INS' inertial navigation system to correct the movements of the lidar in space.
For my information, does Openpylivox V1.1 also support MD40 IMU data?
Thank you,
Fab

ryan-brazeal-ufl, 2020-11-12:
Hi Fab,

It is my understanding that the MID-40 does NOT possess an IMU within the sensor. The MID-40 user manual makes no mention of an IMU as well as the Livox SDK documentation under the data_types section (https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol#33-point-cloudimu-data-data_type) states that data_type 6 (IMU Data) is only supported by the Horizon, Tele, and Avia sensors.

-RGB

closed 2020-12-04 by ryan-brazeal-ufl