ryan-brazeal-ufl/OpenPyLivox #16: Problem in communication between Lidar and Computer

roya-correct-ai, 2020-12-03:
Hi there,

I have a problem when using the Horizon Lidar outdoor. Accidentally the following errors occur:

*** ERROR: HEARTBEAT ERROR MESSAGE RECEIVED ***

*** ERROR: ABNORMAL STATUS MESSAGE RECEIVED ***

The Lidar works perfectly indoors, even outdoor, most of the time works well, but accidentally those errors have happened and when I restart the lidar, they vanished. 

I would appreciate it if you guide me with this issue, is it something related to the wiring? the outdoor weather is around -1 centigrade and the lidar is feed by a vehicle battery power. When the error occurred I checked the IP address ping and it was connected to the computer.

ryan-brazeal-ufl, 2020-12-04:
Hello,
Is this a consistent problem, or does it only occur on occasion? If a heartbeat error is occurring that is a general indication that the connection between the sensor and your computer is a fault. And by connection I mean any cable, network router, ethernet switch, or the Livox Converter box (if being used). I'm happy to help but I'm leaning towards this being a hardware issue and not an OPL issue (but of course I could be wrong).
-RB

roya-correct-ai, 2020-12-04:
> Hello,
> Is this a consistent problem, or does it only occur on occasion? If a heartbeat error is occurring that is a general indication that the connection between the sensor and your computer is a [...]

Thank you for your answer.
Yes, it happens occasionally. nevertheless, if there is a heartbeat error, is it possible that this fault will be removed after a few seconds by any chance? 
I mean what if I change the OPL such that the connection will be checked not one time, but in a loop until the connection comes back to the normal status, and then the OPL continues its normal job. Do you think there is any chance in this manner?

ryan-brazeal-ufl, 2021-01-17:
Hello,
My understanding is if a heartbeat error occurs the lidar sensor stops transmitting observational data. This would be problematic to manage using a continuous monitoring approach without the risk of losing previously collected data. Certainly not impossible, but likely challenging.
-Ryan

closed 2021-02-15 by ryan-brazeal-ufl