ryan-brazeal-ufl/OpenPyLivox #18: Horizon Real Time Data in CSV format

lookquad, 2021-01-04:
Hi, 

I want real time data from livox horizon in CSV format. Is there a way to get it. I don't want to save in binary and then convert in CSV format. 

Thank You

ryan-brazeal-ufl, 2021-01-12:
That's a tricky one, as the writing to CSV operations are/can be very costly (from a real-time sense). Also, simultaneously writing data to a file that is open for reading by another process can be problematic. I have received similar requests for real-time (at least near real-time) output of point cloud data from OPL by other users. The best approach would be to stream the data over a TCP port, but that's a more complicated solution and perhaps not exactly what you had in mind. I'll start thinking about this more and commit 'something' that hopefully works for you.

lookquad, 2021-01-25:
Thank you. I will be waiting for that.