ryan-brazeal-ufl/OpenPyLivox #19: simple test fail

solosails, 2021-01-12:
Hi,

Thanks for providing this code , I'm hoping I can get it working as Im keen to run on a raspberry pi 4 but using the livox drivers, this has ubuntu compatibility issues meaning I can only run it on a Rpi 3

Anyway, I have tried to run your basic example ...

`import openpylivox as opl
sensor = opl.openpylivox(True)
sensor.auto_connect()
sensor.dataStart_RT()
sensor.saveDataToFile("points.csv",0,10.0)   #collect exactly 10.0 seconds of data
while True:
   if sensor.doneCapturing():
      break
sensor.disconnect()`

But get this return...

`Using computer IP address: 127.0.1.1

   *** Discovered a Livox sensor ***
           Type: Mid-40 
         Serial: 0TFDH9U0062***
          IP(s): 192.168.1.80  (ID: 1)
                 
Attempting to auto-connect to the Livox Mid-40  with S/N: 0TFDH9U0062***
Traceback (most recent call last):
  File "test2.py", line 3, in <module>
    sensor.auto_connect()
  File "/home/woody/.local/lib/python3.6/site-packages/openpylivox/openpylivox.py", line 2219, in auto_connect
    numFound = self.connect(self._computerIP, lidarSensorIPs[0], 0, 0, 0)
  File "/home/woody/.local/lib/python3.6/site-packages/openpylivox/openpylivox.py", line 2079, in connect
    self._cmdSocket.sendto(connect_request, (self._sensorIP, 65000))
OSError: [Errno 22] Invalid argument
`

Any ideas what I'm doing wrong ? Thanks!

Best regards, Andrew

ryan-brazeal-ufl, 2021-01-12:
I suspect it is the **sensor.auto_connect()** line that is the culprit, as the auto-discovered IP address for your computer is returning the loopback address of 127.0.1.1. Try instead to use **sensor.auto_connect("192.168.1.X")**, where **X** is the IP address of your computer on the same network (WAN or WiFi) as the Livox sensor is connected to. Run the command **ifconfig** from terminal to discover the IP address(es) of your computer.

I have had no issues running OPL on a RPi4, but I have only used the Raspberry Pi OS, not a Ubuntu distro, but there shouldn't be any issues with the OS, famous last words ;)

solosails, 2021-01-13:
Hi Ryan,

Thank you for taking the time to get back to me.

So, I have tried adding the local IP but have new errors now....

`Connected to the Livox Mid-40  at IP: 192.168.1.152 (ID: 1)
Traceback (most recent call last):
  File "test2.py", line 4, in <module>
    sensor.dataStart_RT()
  File "/home/woody/.local/lib/python3.6/site-packages/deprecated/classic.py", line 285, in wrapper_function
    return wrapped_(*args_, **kwargs_)
  File "/home/woody/.local/lib/python3.6/site-packages/openpylivox/openpylivox.py", line 2481, in dataStart_RT
    self._dataStart_RT()
  File "/home/woody/.local/lib/python3.6/site-packages/openpylivox/openpylivox.py", line 2451, in _dataStart_RT
    self._captureStream = _dataCaptureThread(self._sensorIP, self._dataSocket, "", 1, 0, 0, 0, self._showMessages, self._format_spaces, self._deviceType)
TypeError: __init__() missing 1 required positional argument: 'deviceType'
`

solosails, 2021-01-13:
Hi Ryan,

I seem to have got it working using the longer example script "livox_controller_demo.py"!

This is awesome work - thank you.

One question, is it possible to record and write the .lvx file, or at least an equivalent that can be later converted into a ROS bag for LOAM playback construction?

Thanks again!

ryan-brazeal-ufl, 2021-01-17:
.LVX file export from OPL is on the project's development list. Hopefully, I can find the time needed to implement this functionality soon.

closed 2021-01-17 by ryan-brazeal-ufl

solosails, 2021-01-18:
Awesome, Thanks Ryan.