ryan-brazeal-ufl/OpenPyLivox #20: csv file cannot be imported into Livox Viewer

msben, 2021-01-13:
Hello author, the csv file I exported through `python livox_controller_demo.py` cannot be opened with Livox Viewer, and the error
>Invalid file format, loading file failed ! 

[image]
This is the data I exported
//gyro_x | gyro_y | gyro_z | acc_x | acc_y | acc_z | time
-- | -- | -- | -- | -- | -- | --
0.002178 | -0.00335 | 0.000459 | -0.01749 | 0.003713 | 1.001062 | 28908.48
-0.00208 | 0.004105 | 0.001524 | -0.01273 | -0.00141 | 1.000329 | 28908.48
-0.00208 | -0.00122 | -0.00167 | -0.02473 | -0.00672 | 0.989892 | 28908.49
-0.00208 | -0.00974 | 0.002589 | -0.02674 | 0.002431 | 0.993463 | 28908.49

ryan-brazeal-ufl, 2021-01-13:
Hi msben,
The csv file(s) produced by OPL have not been designed to be opened within the Livox Viewer. CSV point cloud files can be directly opened in numerous point cloud viewing software (including the open-source CloudCompare software). The CSV IMU files (like the one you included in your post) contain the accelerometer and gyroscope observations from the lidar sensor's internal IMU (only applicable if the lidar sensor being used includes an internal IMU). Hope that helps.
-Ryan

solosails, 2021-01-13:
Hi,

Is there any way to convert to .lvx , or livox viewer readable csv? 

Also, is there a way to convert the original raw bin file to .lvx? This would be really handy in order to use Livox's ROS convertor tool to rosbag for further use in Loam_Livox [https://github.com/hku-mars/loam_livox](url)

ryan-brazeal-ufl, 2021-01-17:
Hi solosails,

A high-school student contributor to OPL recently made an .LVX file reader Python class available under OPL. This functionality allows for existing .LVX files to be read into OPL (see [here](https://github.com/ryan-brazeal-ufl/OpenPyLivox/blob/master/openpylivox/BinaryFileReader.py)). But that reminds me, I still need to properly integrate this .LVX reader class within OPL. However, an .LVX file writer class and functionality is still on the long list of development 'things-to-do'. Hopefully, I will be able to commit some time to this effort soon.

I quickly looked at the exported/converted .CSV file that the Livox-Viewer application produces, and it appears that creating an OPL export function (e.g., .convertBin2LivoxCSV) would be straightforward and shouldn't take much time. No timeline promises, but I will look into creating this function soon.

-Ryan

solosails, 2021-01-18:
Brilliant! Thank you for all your work here!

closed 2021-01-20 by msben