ryan-brazeal-ufl/OpenPyLivox #22: I cannot read the ''.csv'' files generated by the livox_controller_demo.py through any tools

lyy-1997-hub, 2021-01-25:
1. when I used:
import pandas as pd
pc_data = pd.read_csv("livoxpoints.csv", header=None, sep='\t', error_bad_lines=False)
print(pc_data)
But only got:
pandas.errors.ParserError: Error tokenizing data. C error: Buffer overflow caught - possible malformed input file.
2. when I used:
filename = 'livoxpoints.csv'
with open(filename, encoding="UTF-8") as f:
    reader = csv.reader(f)
    header_row = next(reader)
    print(header_row)
    first_row = next(reader)
    print(first_row)
But only got:
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xcd in position 19: invalid continuation byte
3. And I used the CloudCompare to open the 'livoxpoints.csv':
I can only get 7 columns of unrecognized Mistaken Code
So, I don't know what's wrong :（ . Could you help me?Please

lyy-1997-hub, 2021-01-26:
By the way, I used the Horizon. Maybe besides the IP address, are there any other parameters that need to be changed?

lyy-1997-hub, 2021-01-26:
And I changed my computer IP address to 192.168.1.50, then used automatic connection "connected = sensor.auto_connect("192.168.1.50")". It can connect successfully, but the file which I got cannot be opened: (

lyy-1997-hub, 2021-01-26:
These are all the information when I run the livox_controller_demo.py :

Using computer IP address: 192.168.1.50

   *** Discovered a Livox sensor ***
           Type: Horizon
         Serial: 1HDDH320010431
          IP(s): 192.168.1.131 (ID: 1)
                 
Attempting to auto-connect to the Livox Horizon with S/N: 1HDDH320010431
Connected to the Livox Horizon at IP: 192.168.1.131 (ID: 1)
      Computer IP Address:    192.168.1.50
      Sensor IP Address(es):  ['192.168.1.131']
      Data Port Number(s):    ['53393']
      Command Port Number(s): ['53394']
      IMU Port Number(s):     ['53395']

   192.168.1.131     -->     F/W Version: 06.09.0000
   192.168.1.131     -->     Serial # 1HDDH320010431
   192.168.1.131     <--     sent lidar spin up request
   192.168.1.131     -->     lidar is spinning up, please wait...
   192.168.1.131     -->     lidar is ready
   192.168.1.131     <--     sent start data stream request
   192.168.1.131     <--     sent update UTC request
   192.168.1.131     <--     sent dual return lidar mode request
   192.168.1.131     <--     sent start IMU data push request
   192.168.1.131     <--     sent turn on rain/fog suppression request
   192.168.1.131     -->     CAPTURING DATA...
   192.168.1.131     -->     writing real-time data to BINARY file: livoxpoints.csv
   192.168.1.131     -->     closed BINARY file: livoxpoints.csv
                                (points: 7011043 good, 197837 null, 7208880 total)
                                (IMU records: 6126)
   192.168.1.131     <--     sent stop data stream request
   192.168.1.131     <--     sent lidar spin down request
Disconnected from the Livox Horizon at IP: 192.168.1.131

lookquad, 2021-01-26:
i think your data is in binary form, u need to convert it into csv. 
in this line-->  192.168.1.131 --> closed BINARY file: livoxpoints.csv , You data is being written in a file with name " livoxpoints.csv", .csv is not the extension its part of the file name extension can be .bin in your case. 

if u see in livox_controller_demo.py file you will find that there is a line a function call at line number 259 which is 
opl.convertBin2LAS(filePathAndNames[i], deleteBin=True)
Use this to convert your bin file into CSV and then use it in cloudcompare or anywhere else.

lyy-1997-hub, 2021-01-27:
> I think your data is in binary form, u need to convert it into csv.
> in this line: 192.168.1.131 --> closed BINARY file: livoxpoints.csv , You data is being written in a file with name " livoxpoints.csv". Here .csv is not the extension, its part of the file name, extension can be .bin in your case.
> 
> if u see in livox_controller_demo.py file you will find that there is a function call at line number 177 which is
> `opl.convertBin2LAS(filePathAndName, deleteBin=True)`
> Use this to convert your bin file into CSV and then use it in cloudcompare or anywhere else.

yes,Thanks!

ryan-brazeal-ufl, 2021-02-15:
Thanks for the discussion. The correct solution has now been provided.
-RB

closed 2021-02-15 by ryan-brazeal-ufl