ryan-brazeal-ufl/OpenPyLivox #23: Putting color like livox viewer

lookquad, 2021-01-26:
Hi, 
I am trying to put the colour in point cloud. What are the parameters i can use in OPL to get a colour scheme like livox viewer. 
 For reference livox viewer has a small documentation for the same.
https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-Viewer#1-color-coding-strategy

Thank You!

ryan-brazeal-ufl, 2021-02-15:
Hello,

There are 2 things you need to be aware of when it comes to the colour scheme/style of a point cloud. 

1) A point cloud may contain individual Red, Green, Blue 8-bit integers (e.g., 207, 14, 162) for each point within the dataset. In this case, when viewing the point cloud within a viewing application (e.g., CloudCompare) you want to ensure that True Colour/RGB (most common descriptions used) is set as the colour scheme for the point cloud.

2) Practically all other 'attributes' about a point cloud (e.g., intensity, Z. coordinates, time, etc.) can also be set as the point cloud's colour scheme within a viewing application. In this case, there is only a single number that defines the individual attribute value for each point. Therefore a colour style needs to also be set in order to define what colour(s) to use. Examples of colour styles would be greyscale, 2 colour linear ramp, 3 colour linear ramp, etc.

Currently, OPL does not provide support for assigning individual R,G,B values for each point, but rather replies on the CloudCompare viewing application to generate a desired colour style using the point cloud's intensity attribute (or any other attribute). That being said, it would not require much work to implement a function to generate the Livox-Viewer R,G,B equivalent integers (as defined within the link you sent, thanks), and then update the CSV and LAS export functions to also include the R,G,B values. I'll add this functionality to the 'To-Do List' for OPL.

Thanks,

Ryan

lookquad, 2021-02-23:
> Hello,
>
> There are 2 things you need to be aware of when it comes to the colour scheme/style of a point cloud.
>
> 1. A point cloud may contain individual Red, Green, Blue 8-bit integers (e.g., 207, 14, [...]

Thank You for this explanation!

closed 2021-02-23 by lookquad