ryan-brazeal-ufl/OpenPyLivox #24: Support for Livox Avia

w4rlock999, 2021-02-16:
I am wondering is this going to work in Livox Avia? I am not checking yet and want to know if anyone using this for Livox Avia.

Or any tips to make this project compatible with Avia, any parameter or something I have to make, I would love to be able to contribute to OPL if it is possible too.

ryan-brazeal-ufl, 2021-02-16:
Hi w4rlock999,

The current version of the Livox SDK Communication Protocol is v1.3 (https://github.com/Livox-SDK/Livox-SDK/wiki/Livox-SDK-Communication-Protocol) but I have not (yet) updated OPL to realize all of the new functionality within v1.3. As a result, the latest Mid-70 and Avia sensors do not have full support within OPL, but they do have partial support. For example, OPL does not currently support Data Types 7 and 8, which are for the triple return observations from the Avia (see section 3.3 in the Livox SDK), but dual return observations are supported as they use the same Data Types as the Horizon sensor. 

If you are interested, it would not require a lot of work to at least implement support for Data Types 7 and 8 within OPL. I'd be happy to review and accept a Pull Request (PR) from you if you wanted to develop this OPL functionality. I only have access to a Mid-40, Mid-100, and Horizon sensors plus a Livox Hub, so any testing that you (or others in the community) can perform using OPL with a Tele-15, Mid-70, or Avia sensors would certainly benefit the overall development of OPL.

I hope to find some free time soon to work on the next release of OPL, but it still may be a while before this happens. 

Thanks for your support and interest in OPL.

-RB

hahakid, 2022-08-17:
AVIA and Mid-70 are tested and the data is normal, only rise a unknow in type:
   *** Discovered a Livox sensor ***
           Type: UNKNOWN
         Serial: 3GGDH7T0010135
          IP(s): 192.168.1.135 (ID: 1)

Tele-15 is tested:
Using computer IP address: 192.168.1.51

   *** Discovered a Livox sensor ***
           Type: Tele-15
         Serial: 1PQDH760010272
          IP(s): 192.168.1.172 (ID: 1)
                 
   *** Discovered a Livox sensor ***
           Type: Tele-15
         Serial: 1PQDH760010272
          IP(s): 192.168.1.172 (ID: 1)

is there any clue of how to handling the bin file (both point cloud and IMU), it seams there is a head.
where can we found the format?

approximatenumber, 2022-08-24:
@hahakid hi there. Just played with AVIA and got it supported (both point data + IMU data). I just made dirty changes, but it works, maybe I create PR later. Ping me if you need it. 
BTW, you can convert binary file using from `convertBin2CSV` function:
```
from openpylivox import convertBin2CSV
convertBin2CSV('/tmp/data.bin', deleteBin=True)
```

hahakid, 2022-08-25:
@approximatenumber I will check later, thanks for your kind reply. ROS supported well, but I am writing a demo for non -CS students for real time detection task, so I choice python.

TBRTK, 2022-08-28:
> @hahakid hi there. Just played with AVIA and got it supported (both point data + IMU data). I just made dirty changes, but it works, maybe I create PR later. Ping me if you need it. BTW, you can convert binary file using `convertBin2CSV` function:
> 
> ```
> from openpylivox import convertBin2CSV
> convertBin2CSV('/tmp/data.bin', deleteBin=True)
> ```

Hi, its been a year or so since I worked with this code, but did you succesfully get IMU data out?
Is it raw or fused ?

hahakid, 2022-08-29:
@TBRTK here is a decoded imu frame of 0.1s:
//gyro_x,gyro_y,gyro_z,acc_x,acc_y,acc_z,time
-0.000642,-0.002169,0.000897,-0.004233,0.010864,0.986681,1657.958862
0.005750,-0.002169,0.001962,-0.004142,0.009399,0.975053,1657.963851
0.001489,0.002092,-0.001234,-0.007712,0.022033,1.009843,1657.968876
-0.004903,0.001027,-0.002299,-0.007438,0.014892,0.998674,1657.973847
-0.000642,-0.001104,-0.000168,-0.002952,0.005554,0.968736,1657.978868
0.005750,0.003157,0.001962,-0.003867,0.020202,0.996385,1657.983850
0.001489,0.003157,-0.004429,-0.006522,0.021117,1.013963,1657.988864
0.001489,-0.002169,-0.000168,-0.005607,0.005737,0.976884,1657.993845
0.004685,-0.001104,-0.000168,-0.005240,0.014526,0.981920,1657.998872
0.001489,0.004223,-0.002299,-0.008445,0.024871,1.013963,1658.003843
0.000424,-0.003234,0.004093,-0.009177,0.011779,0.990434,1658.008864
0.003620,0.001027,-0.001234,-0.003592,0.009124,0.970018,1658.013848
0.002554,0.003157,-0.001234,-0.005698,0.022308,0.998216,1658.018865
-0.002772,-0.001104,-0.001234,-0.007895,0.015716,1.003893,1658.023846
-0.000642,-0.000038,0.003027,-0.008536,0.003448,0.979356,1658.028869
0.000424,-0.001104,0.001962,-0.007071,0.021026,0.993547,1658.033847
-0.002772,-0.001104,0.000897,-0.004874,0.020110,1.007829,1658.038869
0.001489,0.002092,0.000897,-0.008079,0.006469,0.983385,1658.043850
0.006815,-0.001104,-0.000168,-0.003776,0.013244,0.976060,1658.048866
0.007881,-0.000038,0.001962,-0.002952,0.018554,1.005632,1658.053845

approximatenumber, 2022-08-29:
> Hi, its been a year or so since I worked with this code, but did you succesfully get IMU data out? Is it raw or fused ?

hi there, it was raw binary data, both for points and IMU, then I successfully converted it to CSV.

hahakid, 2022-09-15:
#40 @approximatenumber @ryan-brazeal-ufl

ayoubft, 2024-09-25:
Hi,
Still having a problem with the avia sensor,
All I get is ' Connected to UNKNOWN at IP:..', but I get 0 when I try to connect it.
Any work arounds for this?
Thanks