ryan-brazeal-ufl/OpenPyLivox #27: Connection issue with  _heartbeat failure

luc-girod, 2021-03-24:
Hei,

So, I have a LIVOX Horizon connected to a Raspberry Pi 3B, and I am having the copied issue bellow. It worksfor a while but crashes after a while, and rebooting the pi just brings back to it. The way the LIVOX is deployed means I can't unplug the power supply as will anymore (we are working on making this possible in the future).

`Using computer IP address: 192.168.13.35

   *** Discovered a Livox sensor ***
           Type: Horizon
         Serial: 3WEDH760010504
          IP(s): 192.168.13.104 (ID: 1)
                 
Attempting to auto-connect to the Livox Horizon with S/N: 3WEDH760010504

Connected to the Livox Horizon at IP: 192.168.13.104 (ID: 1)
Not connected to Livox sensor at IP: 192.168.13.104
Not connected to Livox sensor at IP: 192.168.13.104
Traceback (most recent call last):
  File "/home/livoxpi/git/SnowyOwl/appV2/acquisition.py", line 99, in <module>
    IP_computer=config.get('acquisition', 'computer_IP'))
  File "/home/livoxpi/git/SnowyOwl/appV2/acquisition.py", line 38, in acquire_clouds
    sensor.lidarSpinUp()                # Data Acquisition
  File "/home/livoxpi/git/OpenPyLivox/openpylivox/openpylivox.py", line 2328, in lidarSpinUp
    states.append(self._heartbeat.work_state)
AttributeError: 'NoneType' object has no attribute 'work_state'
`

ryan-brazeal-ufl, 2021-04-23:
Hi Luc,

Apologies for the late reply. Are you still experiencing this issue? I'm happy to help but there are lots of 'moving parts' within the connection, heart beat, send, receive, communication approach used by the Livox sensors that can be potentially causing the issue. Probably easiest to troubleshoot in real-time as opposed to messaging back and forth.

Let me know,
-Ryan

luc-girod, 2021-05-05:
Hei Ryan,

The LIVOX is now attached 8m in the air on a big pole in the Norwegian mountains. We are still having this issue, and our workaround is to have the raspberry pie attached to it reboot every 30 minutes to reinitialize the connection. Brutal, but, for now, functional. Not sure when I'll have direct access to it again to do debug without the risk of braking the experiment...

Cheers!

closed 2021-07-21 by ryan-brazeal-ufl