ryan-brazeal-ufl/OpenPyLivox #28: BinaryFileReader 'utf-8' codec can't decode

janboehm, 2021-05-24:
Hi,
I understand BinaryFileReader is a 3rd party contribution, maybe someone knows the answer. I'm trying to open a .lvx file using the BinaryFileReader following the example code at the end of the class. It fails with an utf-8 related error. I am using Windows and the file was recorded on Windows. Is there a restriction to certain versions of .lvx or a Linux/Windows issue?

Thanks!

Error:
```
Traceback (most recent call last):
  File "c:/Users/.../plot_lvx.py", line 9, in <module>    test2 = binreaders.lvxreader(livoxfilelocation, 1)
  File "C:\ProgramData\Anaconda3\lib\site-packages\openpylivox\BinaryFileReader.py", line 791, in lvxreader
    diblock[0] = diblock[0].decode('utf-8')
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb2 in position 1: invalid start byte

```
Code:
```
import openpylivox.BinaryFileReader
binreaders = openpylivox.BinaryFileReader.BinaryReaders
livoxfilelocation = './data/2021-02-24 17-45-44.lvx'

test2 = binreaders.lvxreader(livoxfilelocation, 1)

```

janboehm, 2021-05-28:
Upon closer inspection of the code it appears that `lvxreader` can handle only LVX files V1.1.0.0. The LVX files V1.0.0.0 (MID40 prior to FW 3.7) do not contain a  Private Header Block and **cannot** be parsed. I suggest to state this in the comments / documentation. The code could potentially handle it more gracefully by evaluating the version/sub-version values.

closed 2021-05-28 by janboehm

ryan-brazeal-ufl, 2021-05-28:
Hi Dr. Boehm,
First off, my apologies for not getting into your issue sooner. As you noticed, currently `lvxreader` is only 'loosely' integrated within OPL, and that's entirely my fault. The 3rd party contributor graciously offered her code to the project and it's been on my list of things to do for too long. Unfortunately, I still don't have the needed time to get the next update of OPL committed, but it is still an active project.
Thanks for your detailed analysis of your problem and for pinpointing the underlying issue. Full support (read and write) of LVX files will be implemented within OPL at some point.
Hope all this well,
Ryan