ryan-brazeal-ufl/OpenPyLivox #29: Issue While running the demo file  (An unknown error occurred while converting OPL binary data)

pullurinikhil, 2021-06-23:
Hi

I have a LIVOX mid - 40 connected to a Raspberry Pi 4 using a router, after a successful connection, cloned the code installed the necessary lib then ran the demo file.

it got the below issue after running the file(livox_controller_demo). Please find below the error. 

%Run Test_one.py
Connected to the Livox Mid-40  at IP: 192.168.1.80 (ID: 1)
      Computer IP Address:    192.168.1.195
      Sensor IP Address(es):  ['192.168.1.80']
      Data Port Number(s):    ['60001']
      Command Port Number(s): ['50001']
   192.168.1.80      -->     F/W Version: 03.08.0000
   192.168.1.80      -->     Serial # 0TFDHAF00676G4
   192.168.1.80      <--     sent lidar spin up request
   192.168.1.80      -->     lidar is ready
   192.168.1.80      <--     sent start data stream request
   192.168.1.80      <--     sent update UTC request
   192.168.1.80      <--     sent dual return lidar mode request
   192.168.1.80      <--     sent start IMU data push request
   192.168.1.80      <--     sent turn on rain/fog suppression request
   192.168.1.80      -->     CAPTURING DATA...
   192.168.1.80      -->     writing real-time data to BINARY file: test.bin
   192.168.1.80      -->     closed BINARY file: test.bin
                                (points: 413347 good, 86353 null, 499700 total)
   192.168.1.80      <--     sent stop data stream request
   192.168.1.80      <--     sent lidar spin down request
Disconnected from the Livox Mid-40  at IP: 192.168.1.80

CONVERTING OPL BINARY DATA, PLEASE WAIT...
   :  98%|█████████▊| 406012/413347 [00:05<00:00, 83352.42 pts/s]*** ERROR: An unknown error occurred while converting OPL binary data ***
   : 100%|██████████| 413347/413347 [00:19<00:00, 83352.42 pts/s]

How can the issue be resolved?

luc-girod, 2021-07-02:
Hei!

I had the same issue, seems to be that the compatibility with laspy was broken with Laspy 2.0. You need to downgrade to 1.7.0 for it to work.

Some rewrite of openpylivox.py around line 3800 (function _convertBin2LAS ) would be needed.

ryan-brazeal-ufl, 2021-07-21:
Thanks for the discussion and to Luc for the patch to fix the current laspy v2.0 issue. This laspy issue is plaguing several of my projects and a solution will be committed soon.

closed 2021-07-21 by ryan-brazeal-ufl