ryan-brazeal-ufl/OpenPyLivox #30: GPS+PPS Time Sync

adhipermana, 2021-07-08:
Hai Rian

I have a problem when using the openpylivox program when I want to synchronize UTC time against my lidar livox horizon data.
Can you help me how to synchronize this UTC time ?, i use GPS data from the BX305 GNSS receiver and retrieve the GPRMC serial data through my computer, how do I add this PGSS data information to the lidar data using this openpylivox program?

Thanks

adhipermana, 2021-07-08:
sorry, *how do i add this GPS data

lyy-1997-hub, 2021-07-12:
hi, I also want to  use this method, but I cannot find any clue in this project. : (

ryan-brazeal-ufl, 2021-07-21:
Hello,
Thanks for raising this issue/question. I'll have to dig into the specifics on what timing information is supported by the Horizon (and other Livox sensors) as I know the Mid-40 does not support absolute timing information in the form of a GPRMC message directly (i.e., not using a Livox Hub) and rather only supports the relative PPS RS485 timing signal. I am aware that support for external timing signals within OPL is currently missing, but I will improve support for this within upcoming OPL commits.
Thanks,
Ryan