ryan-brazeal-ufl/OpenPyLivox #31: How  can this project to recognize the rising edge of PPS?

lyy-1997-hub, 2021-07-12:
hi, I want to recognize the rising edge of PPS, and when the rising edge is comming, I will put the current GPS time into timestamp. Can anybody help me? thx

ryan-brazeal-ufl, 2021-07-21:
I just posted a reply in the previous issue regarding GPS PPS timing information. But, my first question is, have you properly connected your Livox sensor (which sensor are you using?) to a GPS PPS signal?

lyy-1997-hub, 2021-07-21:
Hello,
Thanks for your reply! My device is the Mid-40. Recently,I have met many difficulties, and I will tell you later. I'm so happy to hear the newest OPL will be coming soon, may I know the accurate release date? I'm really really really need your help！
Thanks,
Ping

发自我的iPhone