ryan-brazeal-ufl/OpenPyLivox #32: Error on demo.py

ManChrys, 2021-07-17:
Hello guys,

when im running the demo script i have the followin error!! Any help would be appreciated <3

geo@geo-V530-15ICB:~/OpenPyLivox$ python livox_controller_demo.py 

Using computer IP address: XXXXXXXx

   *** Discovered a Livox sensor ***
           Type: Mid-40 
         Serial: XXXXXXXXXXXXXXXX
          IP(s): XXXXXX  (ID: 1)
                 
Attempting to auto-connect to the Livox Mid-40  with S/N: XXXXXXXXXXxx
Traceback (most recent call last):
  File "livox_controller_demo.py", line 264, in <module>
    singleSensorDemo()
  File "livox_controller_demo.py", line 46, in singleSensorDemo
    connected = sensor.auto_connect()
  File "/home/geo/OpenPyLivox/openpylivox/openpylivox.py", line 2219, in auto_connect
    numFound = self.connect(self._computerIP, lidarSensorIPs[0], 0, 0, 0)
  File "/home/geo/OpenPyLivox/openpylivox/openpylivox.py", line 2079, in connect
    self._cmdSocket.sendto(connect_request, (self._sensorIP, 65000))
OSError: [Errno 22] Invalid argument

ryan-brazeal-ufl, 2021-07-21:
Hello,
Can you provide any further information on this? Can you adjust the demo script to attempt a manual connection vs. the auto-connection?
Happy to help, just need some more information.
Ryan

ManChrys, 2021-07-26:
Hello @ryan-brazeal-ufl !!

Sorry for the late reply. I tried the manual connection and i had the following result.

[image]

ZirongChan, 2025-06-01:
> Hello [@ryan-brazeal-ufl](https://github.com/ryan-brazeal-ufl) !!
> 
> Sorry for the late reply. I tried the manual connection and i had the following result.
> 
> [image]

hi, it's been a long time haha. but i'm wondering if you have solved the problem about converting binary data to LAS format.